moveit/moveit_tutorials #842: Safety distance for custom made gripper

Cody-Vu, 2025-02-08:
Hello everyone,

I tried to create a new moveit_config package with the moveit setup assistant for the Franka panda and a custom made gripper. I let the default value for the safety distance at 0.03. When I change the value, atleast from what is shown it had no effect (top picture). In comparison (bottom picture) with the default franka gripper it shows a wider range of safety distance, also when I change the value I see can see the change in the setup assistance. Is this just a visual bug because my STL file is to complex?

Collision visualization with the custom made gripper:
[image]

Collision visualization with the franka default gripper:
[image]

rhaschke, 2025-02-09:
> I let the default value for the safety distance at 0.03.

Please clarify, which parameter you refer to. AFAIK there is no parameter `safety distance` in MSA.
What you visualized is just the [collision model of the robot](https://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/urdf_srdf/urdf_srdf_tutorial.html#collision-checking).

Cody-Vu, 2025-02-09:
> Please clarify, which parameter you refer to. AFAIK there is no parameter `safety distance` in MSA. What you visualized is just the [collision model of the [...]

Sorry, I refer to the parameter in the [franka_robot.xacro](https://github.com/frankaemika/franka_ros/blob/c11f000c2737acc22ed8dfeff40555b2644a4294/franka_description/robots/common/franka_robot.xacro#L18C11-L18C33) file in the description package. 
"[safety_distance](https://github.com/frankaemika/franka_ros/blob/c11f000c2737acc22ed8dfeff40555b2644a4294/franka_description/robots/common/franka_hand.xacro#L3C8-L3C142): Minimum safety distance in [m] by which the collision volumes are expanded and which is enforced during robot motions" 

I assume my mesh file is to detailed for the collsion checking (as it's described in this [section](https://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/urdf_srdf/urdf_srdf_tutorial.html#collision-checking). I'll try to display it more simple.

rhaschke, 2025-02-09:
> I refer to the parameter in the [franka_robot.xacro](https://github.com/frankaemika/franka_ros/blob/c11f000c2737acc22ed8dfeff40555b2644a4294/franka_description/robots/common/franka_robot.xacro#L18C11-L18C33) file in the description package.

This parameter is specific to the Franka robot description and just increases the radius of all spheres in its collision model.
It does not automatically apply to your custom gripper mesh. Approximate your gripper mesh with collision spheres the spheres the same way the original gripper is approximated!

In any case, this is not an issue with MoveIt or MoveIt tutorials. Closing.

closed 2025-02-09 by rhaschke